GSI-CS-CO/bel_projects #2: PEXARIA - Selected FPGA in Quartus project is still C5

dslavineccsl, 2016-07-19:
I noticed that for PEXARIA Quartus project (master, proposed_master) there is still C5 FPGA version selected. But for EXPLODER5 I3 is selected. Is this intentional? Should not both be I3?

terpstra, 2016-07-19:
The original pexaria was C5 and the design used to support both. I guess
this can be changed by now.

alex-cscotg, 2016-07-19:
Please add Cesar (bradomyn) to this issue. He wanted to change this.

bradomyn, 2016-07-20:
I have tested it and Quartus doesn't complain and in the report of TimeQuest everything looks OK...
We have to make a stability long test  this in the Testing Facility and check if it affects WR.

dietrichb, 2016-12-05:
Update: I have checked the image built on 29 November for pci_control. It still claims to be C5N.

eb-info dev/wbm0
Project     : pci_control
Platform    : pexaria5 +db[12] +wrex1
FPGA model  : Arria V (5AGXMA3D4F27C5)
Source info : balloon-1309
Build type  : developer preview
Build date  : Tue Nov 29 03:58:46 CET 2016
Prepared by : Jenkins Nightly Build <csco-tg@gsi.de>
Prepared on : tsl002.acc.gsi.de
OS version  : Debian GNU/Linux 8.6 (jessie), kernel 3.16.0-4-amd64
Quartus     : Version 16.0.0 Build 211 04/27/2016 SJ Standard Edition

bradomyn, 2016-12-13:
changed and it is going to be tested. 

BTW the schematics in the wiki  are the old version with the old FPGA 5AGXMA3D4F27C5N

dietrichb, 2016-12-14:
On 12/13/2016 05:00 PM, bradomyn wrote:
> BTW the schematics in the wiki are the old version with the old FPGA
> 5AGXMA3D4F27C5N

yes, the FPGA in the schematics is the C5N
no, the schematics is the most recent one

The FPGA type should be modified in the next revision of the schematics, 
which is not yet planned.

alex-cscotg, 2018-04-20:
Cleaning up old issues:

FPGA (which is placed on PCB): 5AGXM3D4F 27 I3N

bel_projects/syn/gsi_pexarria5/control> $ cat Manifest.py 

syn_device = "5agxma3d4f"
syn_grade = "i3"
syn_package = "27"

I checked exploder5, microTCA, pmc and everything which is based on pexarria5.

=> Everything is fine

Closed

closed 2018-04-20 by alex-cscotg