GSI-CS-CO/bel_projects #3: pexaria: reset controller might be broken

dietrichb, 2016-07-19:
Hi,

I had an issue with a pexaria flashed to an image built on June 22. After flashing, I did not succeed in reloading the image from the flash by writing the magic word to the flash controller (yes, I realized its address had been changed). Only a real power-cycle then helped.

Please verity that the reset controller is working properly.

bradomyn, 2016-07-19:
> I had an issue with a pexaria flashed to an image built on June 22. After
flashing, I did not succeed in reloading the image from the flash by

writing the magic word to the flash controller? You mean reset controller,
dont you?

>Yes, I realized its address had been changed

Can you reproduce it??
> Please verity that the reset controller is working properly.

Please attach the rpd file

dietrichb, 2016-07-20:
- Yes, of course I meant "reset controller" ...

- I did not try

- [rpd.zip](https://github.com/GSI-CS-CO/bel_projects/files/373011/rpd.zip)

The main message is: An item "test reset controller" should be included into the checklist for the release.

bradomyn, 2016-07-22:
Tested, in Exploder5 and SCU it works. Since the problem was detected in a Data Master Project, I think that it should be checked in a DM project with an up-to-date firmware, could you test it, Mathias?

bradomyn, 2016-08-03:
with the update of the gateware in the DM is fixed

closed 2016-08-03 by bradomyn