GSI-CS-CO/bel_projects #7: Exploder - false path setting in .sdc file does not match monster

dslavineccsl, 2016-09-26:
I have noticed that there is a difference between Exploder and Pexaria sdc files for flash clock setting, index 3 vs 4.

EXPLODER
`# cut: wb sys <=> wb flash   (different frequencies and using xwb_clock_crossing)
set_false_path -from [get_clocks {main|\sys_a5:sys_inst|*|general[0].*}] -to [get_clocks {main|\sys_a5:sys_inst|*|general[3].*}]
set_false_path -from [get_clocks {main|\sys_a5:sys_inst|*|general[3].*}] -to [get_clocks {main|\sys_a5:sys_inst|*|general[0].*}]`

PEXARIA
`# cut: wb sys <=> wb flash   (different frequencies and using xwb_clock_crossing)
set_false_path -from [get_clocks {main|\sys_a5:sys_inst|*|general[0].*}] -to [get_clocks {main|\sys_a5:sys_inst|*|general[4].*}]
set_false_path -from [get_clocks {main|\sys_a5:sys_inst|*|general[4].*}] -to [get_clocks {main|\sys_a5:sys_inst|*|general[0].*}]`

Index 4 is correct since it matches the state of clock connections in monster.

alex-cscotg, 2016-09-26:
Thanks, I will check this.

alex-cscotg, 2016-09-26:
Fixed SDC file (for exploder5 and ftm) and tested generated bitstream.

alex-cscotg, 2016-09-26:
Pushed to Balloon.

closed 2016-09-26 by alex-cscotg