GSI-CS-CO/bel_projects #10: [Balloon] eca tools in standard build

dietrichb, 2016-11-10:
When building the project, using "make" without using any other parameters will also build the eca tools.

Those eca tools are intended for use with the old ECA1, but "balloon" is a release focused around the new ECA2.

Suggest to remove "eca" from the list of default components.

alex-cscotg, 2016-11-10:
Maybe we should also remove the "tlu" tools? I'm not sure if EE is still using this component.

Latching inputs can be done via ECAv2 as well.

dietrichb, 2016-11-16:
I think I cleaned this up. See branch "tools_dietrich-2016-11-11"

alex-cscotg, 2016-11-16:
Looks fine and works for me.

Should this be merged into balloon (or into our next release)? @bradomyn

dietrichb, 2016-11-16:
Not yet. Still fixing some other things.

dietrichb, 2016-11-18:
Done. Waiting for merge into balloon --> https://github.com/GSI-CS-CO/bel_projects/pull/12

alex-cscotg, 2016-11-18:
Done.

closed 2016-11-18 by alex-cscotg