GSI-CS-CO/bel_projects #15: [Balloon] Exploder5a master mode doesn't work

alex-cscotg, 2016-11-24:
Log:

```
eb-console dev/wbm0
wrc# mode master
PTP stop
Locking PLL...
Lock timeout.
Command "mode": error -116
wrc# 
```

```
wrc# gui
WR PTP Core Sync Monitor v 1.0
Esc = exit

TAI Time:                  Thu, Jan 1, 1970, 01:21:10
```

I also noticed that the "track phase" LED goes green and the PPS LED starts blinking anyway.

firmware:
```
$ eb-info dev/wbm0
Project     : exploder5_csco_tr
Platform    : exploder5 +db[12] +wrex2
FPGA model  : Arria V (5AGXMA3D4F27I3)
Source info : balloon-1299
Build type  : developer preview
Build date  : Thu Nov 24 03:17:33 CET 2016
Prepared by : Jenkins Nightly Build <csco-tg@gsi.de>
Prepared on : tsl002.acc.gsi.de
OS version  : Debian GNU/Linux 8.6 (jessie), kernel 3.16.0-4-amd64
Quartus     : Version 16.0.0 Build 211 04/27/2016 SJ Standard Edition

  4687f41 scu: fixed timing constraints for both scu platforms
  9b40bbf scu: get rid of nPCI_RESET clk error message
  fa1c31a Merge branch 'tools_dietrich-2016-11-11' into balloon
  ea0dc62 monitoring: added .gitignore file for wr-mon binary
  52293d7 monitoring: fixing minor issues
```

bradomyn, 2016-11-24:
> Assigned #15 <https://github.com/GSI-CS-CO/bel_projects/issues/15> to
> @bradomyn <https://github.com/bradomyn>.

Thanks for the report!

A-Rubini got an invitation to join the Timing Repo, we'll assign him this
issues as a welcome present :)

bradomyn, 2017-03-23:
Not anymore present in Balloon. a lot things has changed in the wrpc-sw and the bug has been fixed.

closed 2017-03-23 by bradomyn