GSI-CS-CO/bel_projects #18: Big offset between WR and ECA PPS

bradomyn, 2016-12-14:
we have observed a big offset between the PSSs of WR and the ones triggered by ECA, around 200ns
It is clear that the ECA need some time to process the events, but what we see, is it normal?

Alex's theory is that we have increased the output delay of the ECA when we have fixed timing violations problems reducing the fan-out... 

For Release 10 we should check:
- delay in ECA
- differences in the output delay increasing the fan-out

alex-cscotg, 2016-12-14:
If I remember correctly we saw 30ns between ECA PPS and WR PPS and 200ns between Switch and nodes?
I will check this again.

alex-cscotg, 2016-12-14:
Okay I measured it again:

Node PPS WR vs Switch: ~110ns
Node PPS ECA vs Switch: ~ 140ns
Node PPS ECA vs Node PPS  WR: ~ 30ns

ECA -> WR -> Switch (ECA PPS comes first, Switch PPS comes last)

bradomyn, 2016-12-14:
my bad, sorry.

Thanks for the measurements

alex-cscotg, 2017-08-01:
Out-dated -> Closed.

closed 2017-08-01 by alex-cscotg