GSI-CS-CO/bel_projects #19: Function Generator: Problem when simultaneously triggering >7 channels

dominicday, 2016-12-15:
**Problem:**
Fault when 12 armed function generator channels are triggered by the same tag.
SCU reboots, in the log:
Unhandled etherbone exception in MSI handler for 0xcf8: Cycle::close: one or more operations failed

**Test environment:**
scuxl0104
timing-rte-RC8-balloon_0

[root@scuxl0104 ~]# saft-ctl baseboard -i
saftlib source version                  : saftlib 1.0.8 (v1.0.8-40-g4734cda): Oct 27 2016 10:20:55
saftlib build info                      : built by cprados on Nov  8 2016 11:41:36 with asl740.acc.gsi.de running CentOS Linux release 7.2.1511 (Core)
[root@scuxl0104 ~]# eb-info dev/wbm0
Project     : scu_control
Platform    : scu3 +comexpress
FPGA model  : Arria II GX (EP2AGX125EF29C5)
Source info : balloon-1282
Build type  : developer preview
Build date  : Thu Nov 17 04:48:56 CET 2016
Prepared by : Jenkins Nightly Build <csco-tg@gsi.de>
Prepared on : tsl002.acc.gsi.de
OS version  : Debian GNU/Linux 8.6 (jessie), kernel 3.16.0-4-amd64
Quartus     : Version 16.0.0 Build 211 04/27/2016 SJ Standard Edition

**Test Data:**
8 coefficients

0 0 0 0 167772 4
0 0 35 0 167772 6
0 0 35 0 727772 6
0 0 35 0 1287772 6
0 0 35 0 1847772 6
0 0 35 0 2407772 6
0 0 35 0 2967772 6
0 0 35 0 3527772 6

/opt/fesa/nfs/local/scuxl0101/fgprofiling/values.ramp

also e.g. values.800 with 100 repeats

**How to reproduce:**

Start 11 copies of fg-ctl set to wait on a tag:
`for FG in 3-0 3-1 4-0 4-1 5-0 5-1 6-0 6-1 8-0 8-1 9-0` 
`do (fg-ctl -t1234 -f fg-$FG < values.ramp & ) ; done`

Start 1 copy of fg-ctl set to generate the same tag:
`fg-ctl -t1234 -g -f fg-9-1 < values.ramp`

(/opt/fesa/nfs/local/scuxl0101/fgprofiling/fg-tag-test.sh)

Using different subsets of FG channels totalling >7 also causes the problem.

alex-cscotg, 2017-08-01:
@stefanrauch @bradomyn @miree @dominicday 

Can we close this issue?

alex-cscotg, 2017-08-07:
Fixed

closed 2017-08-07 by alex-cscotg