GSI-CS-CO/bel_projects #25: VETAR with "balloon" release fails at  GSI_TM_LATCH_V2 on CES RIO4

jamusch, 2017-03-13:
# 1) Description of environment:
________
## Hardware: 
CES RIO4 8070/72 VME with VETAR board
_______
## System: 
R4L-3 adamczew > uname -a
Linux R4L-3 2.6.33.7-shl-3.3-up #1 PREEMPT Wed Oct 5 08:23:43 CEST 2016 ppc ppc ppc GNU/Linux
(Sugarhat (3.3.10)) 

_____________
## Wishbone info:
R4L-3 adamczew > eb-ls dev/wbm0
BusPath        VendorID         Product   BaseAddress(Hex)  Description
1              ---
2              ---
3              ---
4              ---
5              ---
6              ---
7              ---
8              0000000000000651:8752bf45          7ffffff0  ECA_UNIT:EVENTS_IN 
9              0000000000000651:eef0b198           4000000  WB4-Bridge-GSI     
9.1            ---
9.2            0000000000000651:2d39fa8b           4000800  GSI:BUILD_ID ROM   
9.3            0000000000000651:b6232cd3           4000000  GSI:WATCHDOG_MUTEX 
9.4            0000000000000651:fab0bdd8           4001000  GSI:MSI_MAILBOX    
9.5            0000000000000651:5cf12a1c           5000000  SPI-FLASH-16M-MMAP 
9.6            0000000000000651:3a362063           4000010  FPGA_RESET         
9.7            0000000000000651:eef0b198           4040000  WB4-Bridge-GSI     
9.7.1          000000000000ce42:66cfeb52           4040000  WB4-BlockRAM       
9.7.2          0000000000000651:eef0b198           4060000  WB4-Bridge-GSI     
9.7.2.1        000000000000ce42:ab28633a           4060000  WR-Mini-NIC        
9.7.2.2        000000000000ce42:650c2d4f           4060100  WR-Endpoint        
9.7.2.3        000000000000ce42:65158dc0           4060200  WR-Soft-PLL        
9.7.2.4        000000000000ce42:de0d8ced           4060300  WR-PPS-Generator   
9.7.2.5        000000000000ce42:ff07fc47           4060400  WR-Periph-Syscon   
9.7.2.6        000000000000ce42:e2d13d04           4060500  WR-Periph-UART     
9.7.2.7        000000000000ce42:779c5443           4060600  WR-Periph-1Wire    
9.7.2.8        0000000000000651:68202b22           4060700  Etherbone-Config   
9.8            0000000000000651:00000815           6000000  Etherbone_Master   
9.9            0000000000000651:10051981           4000100  GSI_TM_LATCH_V2    
9.10           0000000000000651:b2afc251           4000200  ECA_UNIT:CONTROL   
9.11           0000000000000651:d5a3faea           4000040  ECA_UNIT:QUEUE     
9.12           0000000000000651:7c82afbc           4000020  ECA_UNIT:TLU       
9.13           0000000000000651:18415778           4000080  ECA ACTCHN WBM     
9.14           0000000000000651:d5a3faea           40000c0  ECA_UNIT:QUEUE     
9.15           0000000000000651:5f3eaf43           4000300  wb_serdes_clk_gen  
9.16           0000000000000651:10c05791           4010000  IO_CONTROL         
9.17           0000000000000651:10041000           4080000  LM32-CB-Cluster    
9.17.1         0000000000000651:10040086           4080000  Cluster-Info-ROM   
9.17.2         ---
9.17.3         0000000000000651:54111351           40a0000  LM32-RAM-User      
9.17.4         ---
9.18           0000000000000651:b77a5045           4000400  SERIAL-LCD-DISPLAY 
9.19           ---
9.20           ---
9.21           ---
9.22           ---
9.23           ---
9.24           ---
9.25           ---
9.26           0000000000000651:22ffee84           4020000  INFO_VME           
9.27           ---
9.28           ---
9.29           ---

R4L-3 adamczew > eb-info dev/wbm0
Project     : vetar2a
Platform    : vetar2a +vetar1db2a +wrex1
[... 205 lines cut ...]
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (2) (0xbad00bad)
-R4L-3  :read_meb   :
-R4L-3  :read_meb   :ERROR>> TLU fifo 3 is empty before time stamp read, stat: 0x200
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (1) (0xbad00bad)
-R4L-3  :read_meb   :reset TLU fifo of white rabbit time stamp module vetar 
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (2) (0xbad00bad)
-R4L-3  :read_meb   :
-R4L-3  :read_meb   :ERROR>> TLU fifo 3 is empty before time stamp read, stat: 0x200
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (1) (0xbad00bad)
-R4L-3  :read_meb   :reset TLU fifo of white rabbit time stamp module vetar 
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (2) (0xbad00bad)
-R4L-3  :read_meb   :
-R4L-3  :read_meb   :ERROR>> TLU fifo 3 is empty before time stamp read, stat: 0x200
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (1) (0xbad00bad)
-R4L-3  :read_meb   :reset TLU fifo of white rabbit time stamp module vetar 
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (2) (0xbad00bad)
-R4L-3  :read_meb   :
-R4L-3  :read_meb   :ERROR>> TLU fifo 3 is empty before time stamp read, stat: 0x200
-R4L-3  :read_meb   :white rabbit failure: invalidate current trigger/event (1) (0xbad00bad)
-R4L-3  :read_meb   :reset TLU fifo of white rabbit time stamp module vetar 

Again, reading and writing via eb_cycle works in principle but the GSI_TM_LATCH_V2 unit fifo does not. 
**This device, however, is currently crucial for the DAQ systems of several NUSTAR experiments and should work properly and please be furtherly maintained.**

For previous asterisk VETAR FPGA firmware, the above errors never occured with the same setup of hardware, kernel module and software.

jamusch, 2017-03-24:
**After updating the vetar vme kernel module:**
To countercheck the issue, our vme driver (see https://subversion.gsi.de/dabc/drivers/vetar/)
has been adjusted to the latest modifications concerning git tags
d5187fef4d33b62d77e1e087d8a0f148150bc5c8o (wishbone module)
and the branch vme-drv (tag36fa80db7ca2e29767e81219330a36b09d33a652,  not yet part of balloon?). 

This would correct for the fact that before we had not provided the MSI functionality yet, that is required for the ECA technique of saftlib. So now our driver is mostly consistent with the other vme driver software distributed with balloon release.

However, the MBS DAQ timestamp readout uses neither ECA/MSI nor interrupts, but the
time latch unit GSI_TM_LATCH_V2. 
So the interrupt handler is registered by the driver (as it was before), but is never invoked in the scope of our application (which was also the case with the _asterisk_ firmware on vetar board).
For latency reasons our time stamp fifo readout is always invoked after the DAQ trigger interrupt of the MBS trigger module and is not intended to use an addtional interrupt of the VETAR board.

**So these improvements in our kernel module do not change anything for the problem as described above.**

miree, 2017-08-03:
The issue was caused by me flashing balloon release of the normal vetar2a gateware instead of the vetar2a-ee-butis gateware (which has different IO mapping). The vetar2a-ee-butis gateware was missing in the balloon release. It is now available under the branch balloon_vetar_ee

closed 2017-08-03 by miree