GSI-CS-CO/bel_projects #26: wrpc-sw: change mode p2p -> e2e and back to p2p not working reliable

bradomyn, 2017-03-23:
**Set-up:**
Two Exploder5 connected to each other. One as p2p master the other as p2p slave.

**Firmware WR**
Ver info from the wrpc-sw console:
```
WR Core build: wrpc-v4.0-15-g92e28df (unsupported developer build)
Built: Mar 22 2017 15:15:23
```

**Gateware WR**
```
Project     : exploder5_csco_tr
Platform    : exploder5 +db[12] +wrex2
FPGA model  : Arria V (5AGXMA3D4F27I3)
Source info : zenith-1362
Build type  : developer preview
Build date  : Wed Mar 22 15:15:27 CET 2017
Prepared by : Jenkins Nightly Build <csco-tg@gsi.de>
Prepared on : tsl002.acc.gsi.de
OS version  : Debian GNU/Linux 8.6 (jessie), kernel 3.16.0-4-amd64
Quartus     : Version 16.0.0 Build 211 04/27/2016 SJ Standard Edition

  d159400 monster: add new arriaV phy - update xwr_core
  29a0f50 autogen: fix autogenertated altera ip cores
  79aadb6 submodules: update the submodules wr-cores/wrpc-sw/general-core
  c35a55e monster/top: add SFP cntrl pins to monster and top files of all the formfactors
  e4b2a34 ebm: quick fix for no response bug
```

If the mode is change to e2e in both nodes:]
```mode e2e```

it works and they get synced. If you want to change again the mode to p2p
```mode p2p```

not always the slave synced again, it gets stack in 
```uncalibrated/wr-s-lock```

The config wrpc-sw/ppsi
[config-0320.tar.gz](https://github.com/GSI-CS-CO/bel_projects/files/865788/config-0320.tar.gz)

rubini, 2017-03-23:
This is a known problem with white rabbit initial syntonization.  The timeouts and retries are
wrong, and if the initial exchange of messages is stopped in the middle it may get stuck.
I can add fixing this to my todo.

The same happens if you have massive frame loss during the initial syntonization.
If you want to swtich from one mode to the other, stop ptp on both nodes and start again on both.
When you moved the slave back to p2p, it started syncing. When you moved the master it stopped what it was doing, thus leaving the slave alone in its initial syntonization fsm.

rubini, 2017-04-06:
Timeouts and retries are not wrong, they are just insanely long.  I've seen them several times, but
I can't change them as it's not my area. I'll ask greg whether they are random or well-thought.
We have 15s timeout for locking, and 3 retries, for a total of 1 minute stuck in wr-s-lock
before going back to normal operation. 

Do you remember if the report about being stuck was longer or shorter than one minute?  I imagine
it was shorter, because 1m is a looong time when you test stuff by hand.

I think the bug should be closed, but I'd better wait for ack

alex-cscotg, 2017-08-01:
Out-dated -> Closed.

closed 2017-08-01 by alex-cscotg