GSI-CS-CO/bel_projects #28: PEXARIA gsi time latch unit: fine time word has constant zero bits for BALLOON

jamusch, 2017-03-24:
This problem has been reported and tested by Nikolaus Kurz.
## Situation: 
PEXARIA  on MBS DAQ Linux systems (Debian wheezy). 
Readout of latched timestamps from GSI_TM_LATCH_V2 unit.

**The problem shows up with PEXARIA firmware of release BALLOON
For firmware with branch ASTERISK, everything has proved to work for sure.**

## Problem:
After cold start of system or power cycle, sometimes (but not always) one ore several of the 3 fine time bits of register GSI_TM_LATCH_FIFO_FTSSUB remain constantly 0, independent of the timestamp that is latched afterwards.

## Tests:
Two almost identical systems with a PEXARIA each. After cold start/power cycle the state of the fine time counter bits during subsequent timestamp latch fifo readouts were observed:

1. Test 1:  pexaria 1: bit 0  and bit 1 are always 0; pexaria 2: everything is OK 
1. Test 2:  pexaria 1: OK; pexaria 2: bit 1 is always 0
1. Test 3:  both pexaria are ok
1. Test 4:  both pexaria are ok
1. Test 5:  both pexaria are ok
1. Test 6:  pexaria 1: OK, pexaria 2: bit 1 and bit 2 always 0

**Summary:** 
- both pexaria entities with BALLOON firmware are affected.
- only the least 3 bits of the whole timestamp are affected
- if a pexaria shows this mistake, it remains until next power cycle
- if a pexaria is OK, it remains OK

alex-cscotg, 2017-08-01:
Hi @jamusch 

Could you please check out our latest release candidate?

http://tsl002.acc.gsi.de/releases/c_release_nighty/gateware/ -> vetar2a-ee-butis.rpd 

Does this problem still exist?

Cheers

alex-cscotg, 2018-04-20:
@jamusch : Ist this issue still relevant?

jamusch, 2018-04-20:
This problem has not been seen again after upgrading to cherry release in August 2017.
However, no systematic tests of cold starting the pexaria have been done since. It just did not show up again during regular operation of MBS DAQ tests.

alex-cscotg, 2018-04-20:
I'll leave this issue open until you did a cold start test.

Thx.

closed 2022-07-19 by alexazon