GSI-CS-CO/bel_projects #39: serdes_clk_gen generates frequency before ld_reg_p0_i = '1' when g_selectable_duty_cycle = false

qermit, 2017-07-31:
serdes_clk_gen generates frequency before ld_reg_p0_i = '1' when g_selectable_duty_cycle = false

alex-cscotg, 2017-07-31:
Hi,

what exactly happens on the IO in this case?
Do you need a bug fix?

Cheers

qermit, 2017-07-31:
I tried to use this module with parameters initialized at reset state without wishbone part with constant frequency/period.
I had completely different behavior with and without **g_selectable_duty_cycle** parameter. When g_selectable_duty_cycle was false I have seen no output.
After a moment of reflection I thing there is mistake in both **p_secondary_counter** processes at reset condition. The **per_hi_i** input value should be assigned to **per_count_lo** register.

alex-cscotg, 2017-07-31:
We'll check this.

Thanks for your report.

dietrichb, 2019-08-20:
reproducing, understanding and fixing would be a major effort -> postponed