GSI-CS-CO/bel_projects #47: cherry: remote reset does not reboot host reliably

dietrichb, 2017-08-10:
I have observed a couple of times, that writing the magic word to the RESET controller does not lead to a follow-up reboot of the host system on scuxl0033 (SCU2).

However, it does work on scuxl0143 (SCU3).

I am not sure, if this is a specific issue for scuxl0033 only or if this is an issue common for all SCU2.

alex-cscotg, 2017-08-11:
According to klkaisers last mail:

Could you please test this with another scu2?

klkaiser, 2017-08-11:
My proposal, as soon as Dietrich is back again, we check the reset button of the  blamed SCUXL.
Stucked reset buttons may be the most  probable root cause for this symptom.

Karlheinz Kaiser

Department/Abt.: CSCO
Phone/Telefon:     +49 6159 71 2364
eMail:                      K.Kaiser@gsi.de<mailto:K.Kaiser@gsi.de>

dietrichb, 2017-08-16:
The reset button has been "fixed". However, the inability to reset the host system when the RESET controller is used persists (Just to be sure this is independent of gateware, I also tried a different  gateware (Asterisk), same problem). 

I also tried the reset button on the frontpanel: Neither the FPGA nor the host system "reboot". So I think there is a hardware problem on that specific SCU that is not related to the gateware.

closed 2017-08-16 by dietrichb