GSI-CS-CO/bel_projects #50: TR - Port Down after disable/enable WR port in Switch

bradomyn, 2017-09-05:
When a TR is connected to a WR Switch and the port of the switch, where the TR is connected, is disabled and enabled:

`ifconfig wrX down
ifconfig wrX up`

The WR port of the TR is not going up anymore and the TR needs a power cycle. I could reproduce this problem quite easily (over 60%)

Firmware/Gateware: wr-core/wrpc-sw used in cherry release, I haven't tested the official cherry release

bradomyn, 2017-09-05:
My feeling is that the reset of the Altera GTX is not reacting correctly to the "soft" reset when
`ifconfig wrX down`
`ifconfig wrX up`

I'd start from this point before getting inside of the guts of the Altera GTX which may take >2 weeks of reading plus debugging.

bradomyn, 2017-10-23:
This problem happens only in Arria V TR

alex-cscotg, 2019-07-03:
Commit e6593cb57665a4c3cecdd770e0bb696959d509b6 will fix this for Arria V devices.

Test setup:

- Switch gateware:
```
ABC__XXXm66#wrsw_version
PCB:3.30, FPGA:LX240T; version: v5.0.1-10-gd784723 (hongming); compiled at Dec  7 2018 13:17:41
```

Timing Receiver:
Exploder5a using wrpc-v4.1-7-g56f12bf

dietrichb, 2019-08-20:
this is basically duplicate

closed 2019-08-20 by dietrichb