GSI-CS-CO/bel_projects #126: missing: released gateware with PCIe direct access mapping

dietrichb, 2019-02-06:
FPGA images of the current release should include the direct access mapping for PCIe as this is required for MBS based DAQ systems

alex-cscotg, 2019-02-07:
@miree: Is this not included in doomsday-v4.0.5?

Cloud you please check if this in included in our latest "nightly build"?
http://tsl002.*****************/releases/doomsday-nightly-unstable/gateware/

miree, 2019-02-07:
Yes, it is in doomsday-nightly-unstable. But I made the pull-request after doomsday-v4.0.5 release,
and I didn't remeber about that when we decided to take only Stefan's changes into doomsday-v4.0.6.

alex-cscotg, 2019-02-07:
/closed

Will be inside doomsday-v4.0.7 or next "E" release.

dietrichb, 2019-02-16:
proposal: leave this issue open until the client confirms this is solved

closed 2022-07-19 by alexazon